tobiasschuerg/MH-Z-CO2-Sensors #11: No data read from sensor

Tokn59, 2019-11-19:
Hello Tobias,

Thank you for sharing your project !

I am trying to hget it to work but can't seem to be able to read any data from the sensor. Debug :

14:47:50.895 -> M5Stack initializing...OK
14:47:51.833 -> MHZ 19B
14:47:51.833 -> MHZ: debug mode ENABLED
14:48:03.844 -> -- read CO2 uart ---
14:48:03.844 ->   >> Sending CO2 request............No response after 10 seconds
14:48:06.048 -> -- read CO2 uart ---
14:48:06.048 ->   >> Sending CO2 request............No response after 10 seconds
14:48:08.251 -> -- read CO2 uart ---
14:48:08.251 ->   >> Sending CO2 request............No response after 10 seconds

I see the red led flashing on the sensor every xx seconds so communications seems fine. I tried with and without the PWM Line (used PIN 26) but no result.

Any clues what I might try next ?

Thanks and kind regards,

Tony

tobiasschuerg, 2019-11-19:
I guess theres something wrong with the wireing.

Also to read a value via pwm you need to call `readCO2PWM()` instead of `readCO2Uart()`.

Tokn59, 2019-11-20:
Tobias, I got it to work like this : 

CO2 RX -> T2 RED CONNECTION M5STACK
CO2 TX -> R2 RED CONNECTION M5STACK
CO2 PPM -> PIN 5

In the code : 

// pin for pwm reading
#define PWMPIN  5  // BLUE PIN5 CONNECTION ON M5STACK

MHZ co2(&Serial2, PWMPIN, MHZ19B);

The preheat seems to go on for ever, cancelling with button C returns valid CO2 values.

Thank you for your support !

Tony

tobiasschuerg, 2019-12-27:
You can use just one mode either PWM or UART. 

Which one to use depends on your needs and free pins.
See https://www.arduino.cc/en/tutorial/PWM and https://www.seeedstudio.com/blog/2019/09/25/uart-vs-i2c-vs-spi-communication-protocols-and-uses/

closed 2019-12-27 by tobiasschuerg