tobiasschuerg/MH-Z-CO2-Sensors #13: Sloppy interpretation of response time and preheating time?

AljoschaH, 2019-12-24:
Hi :) 

The code is using the response time of the sensors as a recovery time.  In my understanding the response time T90 is the time needed by the sensors to get a change between 0ppm to 90% of max range. So it does not make sense to wait T90 between two measurements, because it just tells something about the accuracy of a measurement taken, while the ppm concentration is changing.  Close the same, but slightly different is the manner preheating is handled. It does not make sense to wait for an UART request, while the other two sources PWM and ADC are being readed. Like for the response time it just says something about the accurancy of the measurements within this period.

tobiasschuerg, 2019-12-27:
Yes, you might be right.

However I experienced that when not waiting or only waiting for a few seconds, the sensor will stop working after a few days - but maybe that's another isse? Still I got it solved by waiting between the measurements.

So feel free to twiddle the delay and get back with new insights so we can remove or reduce the waiting time.

tobiasschuerg, 2021-01-14:
Closing as there was no new information since more than a year know. 
If anybody comes across this and has new insights, I'm happy to reopen.

closed 2021-01-14 by tobiasschuerg