tobiasschuerg/MH-Z-CO2-Sensors #16: PPMUart && Temperature N/A

quintendewilde, 2020-10-16:
Hi, 

I get this in the serial monitor. 

----- Time from start: 1099 s
PPMuart: n/a, PPMpwm: 634, Temperature: n/a

Running this on a Wemos D1 

With this sketch 

#include <ESP8266WiFi.h>
#include <SoftwareSerial.h>
#include <MHZ.h>

// pin for pwm reading
#define CO2_IN 16

// pin for uart reading
#define MH_Z19_RX D1  // D7
#define MH_Z19_TX D2  // D6

MHZ co2(MH_Z19_RX, MH_Z19_TX, CO2_IN, MHZ19B);

void setup() {
  Serial.begin(9600);
  pinMode(CO2_IN, INPUT);
  delay(100);
  Serial.println("MHZ 19B");

  // enable debug to get addition information
  // co2.setDebug(true);

  if (co2.isPreHeating()) {
    Serial.print("Preheating");
    while (co2.isPreHeating()) {
      Serial.print(".");
      delay(5000);
    }
    Serial.println();
  }
}

void loop() {
  Serial.print("\n----- Time from start: ");
  Serial.print(millis() / 1000);
  Serial.println(" s");

  int ppm_uart = co2.readCO2UART();
  Serial.print("PPMuart: ");

  if (ppm_uart > 0) {
    Serial.print(ppm_uart);
  } else {
    Serial.print("n/a");
  }

  int ppm_pwm = co2.readCO2PWM();
  Serial.print(", PPMpwm: ");
  Serial.print(ppm_pwm);

  int temperature = co2.getLastTemperature();
  Serial.print(", Temperature: ");

  if (temperature > 0) {
    Serial.println(temperature);
  } else {
    Serial.println("n/a");
  }

  Serial.println("\n------------------------------");
  delay(5000);
}

tobiasschuerg, 2020-10-26:
`PPMuart: n/a, PPMpwm: 634, Temperature: n/a`
Seems like reading via pwm worked and reading via uart did not.

You probably need to check if your wiring is correct and if the pins you specified `D1` and `D2` are the right pins for serial.

closed 2020-10-26 by tobiasschuerg

nbarjolin, 2020-11-15:
Hello,
I would like to let you know that I ran into a similar issue with `readCO2UART()` failing because `    int count = _serial->readBytes(response, 9);` was returning -1.

This was due to an outdated version of EspSoftwareSerial. Problem was solved by updating the Arduino core for ESP8266 to the latest version (2.7.4, was 2.6.3 initially).

Best regards,