tobiasschuerg/MH-Z-CO2-Sensors #22: Support for MH-Z19C?

marcelstoer, 2021-03-08:
Wondering if anyone has tested this with the MH-Z19C yet? According to the manufacturer information which we make available at https://thingpulse.com/product/mh-z19-co2-sensor/ it should be a drop-in replacement. So far we've been using https://github.com/WifWaf/MH-Z19.

rams007, 2021-03-20:
I tested.  i have MH-Z19C  its worked .
I just pass MHZ19B as parametr and  its worked.
with PWM i alwayse get data
with UART i get response -5  BUT  if i comment line 108 in MHZ.cpp  it  worked . 
i dont know why  isReady  returned false  but if delete this check -  all working correct.

wollewald, 2021-06-06:
I have made the same experience on a MH-Z19C and on a MH-Z14 (without suffix). Apart from that it works fine.

tobiasschuerg, 2021-09-28:
I don't own that sensor, but we can easily add a configuration for it. As stated above there is probably no big change.

tobiasschuerg, 2021-11-30:
Support for the MH-Z19C is added with the latest PR.