tobiasschuerg/MH-Z-CO2-Sensors #27: Constant in `readCO2PWM` depends on range

v-borovikov, 2021-12-27:
Hi!

Just started working with MH-Z sensors family. 
After reading MH-Z19 sensors datasheet and examing some libraries I've noticed that every one copies exactly same mistake. 

```
ppm_pwm = 2000 * (th - 2) / (th + tl - 4);
          ^^^^--- depends on range                                     
```

Instead of constant `2000` there should be current range. So, if sensor was configured with i.e. `5000` range, this formula will lower the value of measurement.

tobiasschuerg, 2022-02-16:
You're right!

I currently don't have much capacity left - feel free to submit a PR

tobiasschuerg, 2022-02-18:
Initially it was already 5000 ;-)

As  pointed out by @v-borovikov the range will not be the same for everyone.
To address this issue the range (2000 or 5000) should be configurable - ideally it is initially read from the sensor (command `0x99`) and then used...

drleavsy, 2022-02-20:
New PR is https://github.com/tobiasschuerg/MH-Z-CO2-Sensors/pull/34

closed 2022-02-22 by tobiasschuerg