tobiasschuerg/MH-Z-CO2-Sensors #30: only getting "PPMurat: n/a" as output

Boerny41, 2022-01-26:
Hi,

all I am getting as output is 

`17:45:17.352 -> ..................................
17:48:07.348 -> 
17:48:07.348 -> ----- Time from start: 180 s
17:48:08.567 -> PPMuart: n/a`

I have tried every combination of pins I could think of and tried to change them. Besides that the code is unchanged. Controller is a D1 mini.

Is there anything I am doing wrong?

tobiasschuerg, 2022-01-31:
> Is there anything I am doing wrong?

That's not enough information. Actually you are just saying "it's not working" ...

I once also used a Wemos and from a look at some quite old code, I can tell that I used pins `RX_PIN 13` amd ` TX_PIN 15` - but if you really tried *every combination of pins I could think of*  then there will be some other issue. 

Maybe the sensor is broken?

tobiasschuerg, 2022-01-31:
Maybe check out some Blog posts from other people about the correct wireing.
For example https://esp8266-server.de/MQTT/CO2-Sensor.html

Boerny41, 2022-02-01:
I was unable to figure out what was wrong but I still guess something with the pinout. In the end I used esp-easy.

closed 2022-02-01 by Boerny41