tobiasschuerg/MH-Z-CO2-Sensors #39: Call of overloaded 'MHZ(const uint8_t&, const uint8_t&, const int&)' is ambiguous

cdjost, 2022-04-26:
> nodemcuWeatherAtHome:75:37: error: call of overloaded 'MHZ(const uint8_t&, const uint8_t&, const int&)' is ambiguous
>    75 | MHZ co2(MH_Z19_RX, MH_Z19_TX, MHZ19B);
>       |                                     ^
> In file included from /home/christoph/Arduino/nodeMcuWeatherAtHome/src/nodemcuWeatherAtHome/nodemcuWeatherAtHome.ino:17:
> /home/christoph/Arduino/libraries/MH-Z_CO2_Sensors/MHZ.h:36:3: note: candidate: 'MHZ::MHZ(uint8_t, uint8_t, uint16_t)'
>    36 |   MHZ(uint8_t pwmpin, uint8_t type, uint16_t range = RANGE_5K);
>       |   ^~~
> /home/christoph/Arduino/libraries/MH-Z_CO2_Sensors/MHZ.h:35:3: note: candidate: 'MHZ::MHZ(uint8_t, uint8_t, uint8_t)'
>    35 |   MHZ(uint8_t rxpin, uint8_t txpin, uint8_t type);
>       |   ^~~
> exit status 1
> call of overloaded 'MHZ(const uint8_t&, const uint8_t&, const int&)' is ambiguous

Compiler can not distinguish between HZ::MHZ(uint8_t, uint8_t, uint8_t) and MHZ::MHZ(uint8_t, uint8_t, uint16_t)

sisco0, 2022-04-26:
Could you kindly test the following source code line instead?
```c
MHZ::MHZ co2(MH_Z19_RX, MH_Z19_TX, MHZ19B);
```

drleavsy, 2022-04-26:
Can you please provide more details? 
From the first glance, it seems you try to pass incorrect parameters to the class constructor. Have you updated your library via Arduino library manager to 1.3.1? Have you modified your header files in any way? The latest version is fully compatible with the previous version, and user don't need to make any changes, it should work right out-of-the-box without adding additional constructor arguments. (because there is a default last argument for PWM which is not shown during class initialization and in case you want to change it , it has to be or RANGE_2K or RANGE_5K , like so MHZ co2(MH_Z19_RX, MH_Z19_TX, CO2_IN, MHZ19B, RANGE_2K); ).

korintje, 2022-04-30:
I have the same issue like below.

# Code
```
#include <MHZ.h>
#define RX_PIN 16
#define TX_PIN 17
MHZ co2(RX_PIN, TX_PIN, MHZ19C);
```

# Error message
```
mhz19c.h:12:31: error: call of overloaded 'MHZ(int, int, const int&)' is ambiguous
 MHZ co2(RX_PIN, TX_PIN, MHZ19C);
                               ^
In file included from C:\Users\username\Documents\Arduino\mysketch\mhz19c.h\mhz19c.h.ino:1:0:
C:\Users\username\Documents\Arduino\libraries\MH-Z_CO2_Sensors/MHZ.h:36:3: note: candidate: MHZ::MHZ(uint8_t, uint8_t, uint16_t)
   MHZ(uint8_t pwmpin, uint8_t type, uint16_t range = RANGE_5K);
   ^
C:\Users\username\Documents\Arduino\libraries\MH-Z_CO2_Sensors/MHZ.h:35:3: note: candidate: MHZ::MHZ(uint8_t, uint8_t, uint8_t)
   MHZ(uint8_t rxpin, uint8_t txpin, uint8_t type);
   ^
exit status 1
call of overloaded 'MHZ(int, int, const int&)' is ambiguous
```

# Environment
- Arduino IDE version: 1.8.19 (Windows Store 1.8.57.0)
- board: "NodeMCU-32S, 80MHz, 921600"
- OS: MS Windows

# Others
- I have installed "MH-Z CO2 Sensors" library 1.3.1 via Arduino library manager.
- I used the installed library as is, without any modification.
- I have also installed "EspSoftwareSerial" library 6.16.1 via Arduino library manager because `SoftwareSerial.h` is needed.
- When I changed `MHZ` to `MHZ::MHZ`, another error was emitted:
  ```
  mhz19c.h:12:1: error: 'MHZ::MHZ' names the constructor, not the type
   MHZ::MHZ co2(RX_PIN, TX_PIN, MHZ19C);
   ^
  exit status 1
  'MHZ::MHZ' names the constructor, not the type
  ```

**Thank you for developing the useful tool!**

drleavsy, 2022-04-30:
Yep, the bug was reproduced, working on it.

drleavsy, 2022-04-30:
The bug is fixed in this PR: https://github.com/tobiasschuerg/MH-Z-CO2-Sensors/pull/38
@tobiasschuerg can you please review and merge it as soon as you can.

korintje, 2022-05-01:
@drleavsy 
Thank you for the very quick fix!

tobiasschuerg, 2022-05-03:
It's merged , thanks for the fix!

closed 2022-05-03 by tobiasschuerg