tobiasschuerg/MH-Z-CO2-Sensors #45: Version 1.4.1 isn't visible in Platform.IO registry

SNMetamorph, 2023-10-19:
That's just because you missed to create tag for 1.4.1 release

tobiasschuerg, 2023-10-25:
Thanks for the note. 
Just created a release for `1.4.1` and `the new `1.5.0:
https://github.com/tobiasschuerg/MH-Z-CO2-Sensors/releases/tag/1.5.0
Those versions should appear in the registires soon.

closed 2023-10-25 by tobiasschuerg

thehans, 2023-11-01:
From what I have read PlatformIO registry should update existing libraries every 24hrs.  

It seems this has not occurred (latest and only version in PlatformIO registry is 1.4.0 ), so I searched for why this would happen, and found this topic: https://community.platformio.org/t/library-is-not-updating-in-platformio-registry/33198

Where it sounds like the solution is that you must run `pio pkg publish`
https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_publish.html

Has this been done?  I think only repository owner has authorization to do so.

tobiasschuerg, 2023-11-02:
@thehans seems like the crawler found it automatically:
https://registry.platformio.org/libraries/tobiasschuerg/MH-Z%20CO2%20Sensors